adafruit/Adafruit-MLX90614-Library #9: arduino nano every, Wire, and MLX90614

EdgeEdx, 2020-03-31:
- Arduino board:  **Nano Every**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.12**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **compile mlxtest**

issue observed:

The requestFrom call does not seem to match what is specified in Wire package.  Casting the second parameter (the quantity parameter) for requestFrom to a size_t seems to fix the compile issue.

Compiling library "Adafruit_MLX90614_Library"
"C:\\ls\\arduino\\arduino-1.8.12\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\ls\\arduino\\arduino-1.8.12\\portable\\packages\\arduino\\hardware\\megaavr\\1.8.5\\cores\\arduino/api/deprecated" "-IC:\\ls\\arduino\\arduino-1.8.12\\portable\\packages\\arduino\\hardware\\megaavr\\1.8.5\\cores\\arduino" "-IC:\\ls\\arduino\\arduino-1.8.12\\portable\\packages\\arduino\\hardware\\megaavr\\1.8.5\\variants\\nona4809" "-IC:\\ls\\arduino\\arduino-1.8.12\\portable\\packages\\arduino\\hardware\\megaavr\\1.8.5\\libraries\\Wire\\src" "-IC:\\ls\\arduino\\arduino-1.8.12\\portable\\sketchbook\\libraries\\Adafruit_MLX90614_Library" "C:\\ls\\arduino\\arduino-1.8.12\\portable\\sketchbook\\libraries\\Adafruit_MLX90614_Library\\Adafruit_MLX90614.cpp" -o "C:\\Users\\edgeb\\AppData\\Local\\Temp\\arduino_build_303423\\libraries\\Adafruit_MLX90614_Library\\Adafruit_MLX90614.cpp.o"
In file included from C:\ls\arduino\arduino-1.8.12\portable\sketchbook\libraries\Adafruit_MLX90614_Library\Adafruit_MLX90614.h:24:0,

                 from C:\ls\arduino\arduino-1.8.12\portable\sketchbook\libraries\Adafruit_MLX90614_Library\Adafruit_MLX90614.cpp:19:

C:\ls\arduino\arduino-1.8.12\portable\packages\arduino\hardware\megaavr\1.8.5\libraries\Wire\src/Wire.h: In member function 'uint16_t Adafruit_MLX90614::read16(uint8_t)':

C:\ls\arduino\arduino-1.8.12\portable\packages\arduino\hardware\megaavr\1.8.5\libraries\Wire\src/Wire.h:62:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int, int);

             ^~~~~~~~~~~

EdgeEdx, 2020-04-01:
Indeed, the example does compile, link, load and run.  Thank you for investigating.

siddacious, 2020-04-01:
Done!

closed 2020-04-01 by siddacious

apocolipse, 2020-11-07:
I know this is months old and closed, but @EdgeEdx is your MLX90614 sensor working properly on the nano every?  I have 2 nano every boards and 2 mlx90614 sensors, they wont work.  Both sensors work fine with a regular arduino nano.  Googling i've found a few other users on arduino forums with same issues but no resolution.  Also no other library seems to be able to work.  I'm not familiar enough with I2C to debug further, any idea where I should start looking @siddacious?

siddacious, 2020-11-09:
@apocolipse Since another use is able to sue the combination, I think your issue is better suited for the forums. Please make a post at https://forums.adafruit.com explaining  what your're trying to do, what you're expecring, and what actually happened. Please also include clear, well lit, in focus pictures that show the parts and wiring of your setup clearly.

thanks!

DarioSiroki, 2020-11-12:
@apocolipse identical issue here, please let me know if you find out anything