adafruit/Adafruit-MLX90614-Library #11: EEPROM registers address

pcremades, 2020-04-25:
Dear developer,
I think there is an error in the EEPROM registes address definitions. According to the MLX90614 datasheet:
TOmax   0x00
TOmin    0x01
PWMCRTL  0x02
Ta_range 0x03
EmissCoef 0x04
ConfigReg 0x05

But in the Header file of the library these are defined as
TOmax   0x20
TOmin    0x21
PWMCRTL  0x22
Ta_range 0x23
EmissCoef 0x24
ConfigReg 0x25

Am I missing something?

ladyada, 2020-04-25:
[image]

ladyada, 2020-04-25:
see datasheet for explaination

closed 2020-04-25 by ladyada

danial2016, 2020-12-21:
Ye, the datasheet was a bit misguiding concerning the EEPROM register addresses. But do the following to get the real EEPROM address (virtually adding 2 in the upper nibble):
0b0010_0000+ 0b000x_xxxx = 0b001x_xxxx where x_xxxx is the EEPROM reg address at the page OP refers to.

ladyada, 2020-12-21:
its been a long time since we tried this chip, if you can submit a PR indicating the changes to use the EEPROM we'll take a look :)