adafruit/Adafruit-MLX90614-Library #17: M5StickC NCIR Hat has SDA 0, SCL 26. But the defualt SDA(32) and SCL(33) are cannot be changed.

iotmaker-cc, 2020-10-07:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  M5StickC

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.12

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
  
  1) begin() method only call Wire.begin()
      There is no argument and only use defualt SDA and SCL for Wire.begin().
      M5StickC has deafult SDA 32 and SCL 32. 
      But M5StickC NICR Hat uses SDA 0 and SCL 26. 
      So there is a need to set new SDA and SCL.
  2) My recommendation
      
 At Adafruit_MLX90614.h

     #if defined(ESP8266)
    bool begin(int sda, int scl);
    bool begin(int sda, int scl, uint8_t address);
    bool begin();
    #elif defined(ESP32)
    bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); 
    #else 
    bool begin();
    #endif

    At Adafruit_MLX90614.cpp

    #if defined(ESP8266)
    bool Adafruit_MLX90614::begin(int sdaPin, int sclPin) {
        Wire.begin(sdaPin,sclPin);
       return true;
    }

    bool Adafruit_MLX90614::begin(int sdaPin, int sclPin, uint8_t address) {
       Wire.begin(sdaPin,sclPin,address);
       return true;
    }

   bool Adafruit_MLX90614::begin() {
      Wire.begin();
      return true;
   }
   #elif defined(ESP32)
   bool Adafruit_MLX90614::begin(int sdaPin, int sclPin, uint32_t frequency) {
       Wire.begin(sdaPin,sclPin,frequency);
       return true;
  }
  #else
  bool Adafruit_MLX90614::begin() {
     Wire.begin();
     return true;
  }
  #endif

ladyada, 2020-10-07:
please chat with m5stack so they can update their documentation

closed 2020-10-07 by ladyada

iotmaker-cc, 2020-10-07:
My concern is that begin() method needs additional arguments to set new SDA and SCL pin number.

iotmaker-cc, 2020-10-07:
> please chat with m5stack so they can update their documentation

I think it is not the problem on M5stack documentation.

My recommendation  is that your begin() method needs additional arguments to set new SDA and SCL pin number for general purpose.

ladyada, 2020-10-07:
that is not standard usage for Arduino, m5stack needs to update their core to have a default I2C port

iotmaker-cc, 2020-10-07:
> that is not standard usage for Arduino, m5stack needs to update their core to have a default I2C port

I understand what you say. 
In case of ESP32, I2C pins can be allocated dynamically, and can have two I2C pins at the same time. So it is better  your begin() method can choose I2C pins. 
M5StickC has one grove connector (pin no 32,33) and it can be used as default I2C, but it has other pins(pin no 0,26) on the head side that can be used as another I2C. Their NICR hat is attached on the head side, and it uses second I2C. So it is better for your begin() method to have arguments to select I2C pins.

At this time, to solve the problem I can use Wire.begin(0,26) instesd of using your mlx.begin(). 

Regards.

ladyada, 2020-10-07:
you can use setPins
https://github.com/espressif/arduino-esp32/issues/3779