adafruit/Adafruit-MLX90614-Library #20: Multiple I2c Does not working -Always returning high number 1037

fervidautomation, 2020-12-07:
Hello Team, 
I am using Adafruit_MLX90614 for capturing the temperature. When I am using a single I2C sensor it is working fine and returning the correct temperature. 

When I tried to use Adafruit_MLX90614 wih MAX30105.h library. My MAX30102 is working fine and  returning the correct value for MAX30102 sensor but MLX90614 always return 1037 lke below output

@	⸮⸮X⸮h̘⸮1⸮M⸮⸮mlx status is status is mlx Ambient = 
1037.55*C	Object = 1037.55*C
Ambient = 1899.59*F	Object = 1899.59*F
status is mlx Ambient = 
1037.55*C	Object = 1037.55*C
Ambient = 1899.59*F	Object = 1899.59*F
status is mlx Ambient = 
1037.55*C	Object = 1037.55*C
Ambient = 1899.59*F	Object = 1899.59*F
100.00,100.00,80.00,85.00,90.00,95.00,100.00
status is mlx Ambient = 
1037.55*C	Object = 1037.55*C
Ambient = 1899.59*F	Object = 1899.59*F

The same wiring works fine when I am using a single I2C device. Any help will be highly appreciated Thank you.

Code:

/*
   
  ## Hardware Connections (Breakoutboard to ESP32 Arduino):
  -VIN = 3.3V
  -GND = GND
  -SDA = 21 (or SDA)
  -SCL = 22 (or SCL)
  -INT = Not connected

   

*/

#include <Wire.h>
#include <MAX30105.h> //sparkfun MAX3010X library
#include <Adafruit_MLX90614.h>

  bool status; 
Adafruit_MLX90614 mlx = Adafruit_MLX90614(0x5A);
MAX30105 particleSensor;
TwoWire I2CBME = TwoWire(0);
 TwoWire I2CMLX = TwoWire(1);

 
#define USEFIFO
#define I2C_SDA 33  //21
#define I2C_SCL 32  //22

#define I2C_SDA1 21  //21
#define I2C_SCL1 22  //2
void setup()
{
  Serial.begin(9600);
 I2CMLX.begin(I2C_SDA1, I2C_SCL1, I2C_SPEED_FAST);

   mlx.begin();
   I2CBME.begin(I2C_SDA, I2C_SCL, I2C_SPEED_FAST);
 // Serial.println("mlx start Initializing...");

//status = mlx.begin();  
  Serial.print("mlx status is " + status) ;
 // mlx.begin();
 
   if (!mlx.begin()) //Use default I2C port, 400kHz speed
  {
     Serial.println("mlx was not found. . ");
      while (1);
[... 147 lines cut ...]
        Serial.print(","); Serial.print(90.0); //warning SpO2 line
        Serial.print(","); Serial.print(95.0); //safe SpO2 line
        Serial.print(","); Serial.println(100.0); //max SpO2 line
        //#endif

 
Serial.println();
delay(1000);

      }
    }
    if ((i % Num) == 0) {
      double R = (sqrt(sumredrms) / avered) / (sqrt(sumirrms) / aveir);
      // Serial.println(R);
      SpO2 = -23.3 * (R - 0.4) + 100; //http://ww1.microchip.com/downloads/jp/AppNotes/00001525B_JP.pdf
      ESpO2 = FSpO2 * ESpO2 + (1.0 - FSpO2) * SpO2;
      //  Serial.print(SpO2);Serial.print(",");Serial.println(ESpO2);
      sumredrms = 0.0; sumirrms = 0.0; i = 0;
      break;
    }
    particleSensor.nextSample(); //We're finished with this sample so move to next sample
    //Serial.println(SpO2);
  }
#endif
}

ladyada, 2020-12-07:
you need to use setPins() to set the i2c pins on ESP32 because libraries call begin() in the library
https://github.com/espressif/arduino-esp32/issues/3779

fervidautomation, 2020-12-07:
I got below error when I tried to setpins()

32:9: error: 'class TwoWire' has no member named 'setPins'
    Wire.setPins(I2C_SDA1,I2C_SCL1);
         ^
exit status 1
'class TwoWire' has no member named 'setPins'

#include <Wire.h>
#include <Adafruit_MLX90614.h>
 
#include <MAX30105.h> //sparkfun MAX3010X library

#define IR1 0x5A
bool statusMLX;
bool statusMAX;
Adafruit_MLX90614 mlx = Adafruit_MLX90614(IR1);
 MAX30105 maxSensor;
 //TwoWire I2CMAX = TwoWire(0);

   
#define I2C_SDA 33  //21
#define I2C_SCL 32  //22

#define I2C_SDA1 21  //21
#define I2C_SCL1 22  //22

inline const char * const BoolToString(bool b)
{
  return b ? "true" : "false";
}

void setup()
{
Serial.begin(9600);
#ifdef ARDUINO_ARCH_ESP32
   Wire.setPins(I2C_SDA1,I2C_SCL1);
#endif
  
statusMLX= mlx.begin();
//statusMAX =maxSensor.begin(Wire, I2C_SPEED_FAST);
  Serial.println("mlx  status is  ");
  Serial.println( BoolToString( statusMLX)) ;
    Serial.println("MAX  status is  ");
  Serial.println( BoolToString( statusMAX)) ;

  
   I2CMAX.begin(I2C_SDA, I2C_SCL, I2C_SPEED_FAST);
}
 
void loop()
{
 Serial.print("Ambient = ");
Serial.print(mlx.readAmbientTempC());
Serial.print("*C\tObject = ");
Serial.print(mlx.readObjectTempC());
Serial.println("*C");
Serial.print("Ambient = ");
Serial.print(mlx.readAmbientTempF());
Serial.print("*F\tObject = ");
Serial.print(mlx.readObjectTempF());
Serial.println("*F");
 
Serial.println();
delay(1000);
}

ladyada, 2020-12-07:
you'll need to figure out for your platform how to create two different i2c ports without using begin(pin1, pin2) - not sure what you're doing or how

fervidautomation, 2020-12-07:
Thank you for your response.

I am using ESP32 Wroom with 2 I2C sensor. MAX30102 and MLX90614.
 
I need to get value from both sensor after a set interval.  individually both the sensor are working in poc project. When I am trying to access both in single program than MLX90614 is not returning the correct value. it always return 1037.

Thank you in advance for your help

mjohanarif, 2021-01-07:
i am doing similar project like yours, so have you been able to do it? and how did you do it?

caternuson, 2021-06-29:
Closing. This isn't an issue with this library. It's a generic issue with configuring I2C port(s) for use with multiple devices.

closed 2021-06-29 by caternuson