adafruit/Adafruit-MLX90614-Library #21: Why is the MLX90614 Datasheet Measurement Resolution 0.02 C but in after I try, it can be 0.01 C?

jake-winson, 2020-12-15:
I thought I would get even data, but it turns out I can get odd data too.

how it can be happened?

jake-winson, 2020-12-15:
I couldn't provide my example, but here's the example from internet 
[image]

jake-winson, 2020-12-15:
I couldn't provide my example, but here's the example from internet 
[image]

on the last data, the object is 27.16 degree C

ladyada, 2020-12-15:
rounding effect and floating point inconsistancies!
https://github.com/adafruit/Adafruit-MLX90614-Library/blob/master/Adafruit_MLX90614.cpp#L117
https://floating-point-gui.de/

closed 2020-12-15 by ladyada