adafruit/Adafruit-MLX90614-Library #24: Always returns true?

jhihn, 2021-04-30:
https://github.com/adafruit/Adafruit-MLX90614-Library/blob/06b30a8f5663ca899685519face14f80381e1689/Adafruit_MLX90614.cpp#L40

Seems like if the sensor is not attached the begin should fail?
Sparkfun does that:
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library/blob/9dba549951001c3f5994b75c4ea327c4b560bff3/src/SparkFunMLX90614.cpp#L36

However I have found that the resutl of begin() is not a reliable predictor of being able to get a temperature reading.

T365, 2021-05-18:
Mine problem too...

caternuson, 2021-06-29:
Please update to the 2.0.0 release of the library:
https://github.com/adafruit/Adafruit-MLX90614-Library/releases/tag/2.0.0
and try that. The call to `begin()` now conditionally returns `true`/`false` based on successful initing of the sensor.