adafruit/Adafruit-MLX90614-Library #25: Compatibility with Arduino Due

LEBsci, 2021-06-29:
- Arduino board:  Arduino Due

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
```
/*****************************************************
Hardware hookup: 
MLX90614 ------------- Arduino Due
  VDD ------------------ 3.3V
  VSS ------------------ GND
  SDA ------------------ SDA (20)
  SCL ------------------ SCL (21)
********************************************************/

#include <Wire.h>
#include <Adafruit_MLX90614.h>

Adafruit_MLX90614 mlx = Adafruit_MLX90614();

void setup() {
  Serial.begin(9600);

  Serial.println("Adafruit MLX90614 test");  

  mlx.begin();  
}

void loop() {
  Serial.print("Ambient = "); Serial.print(mlx.readAmbientTempC()); 
  Serial.print("*C\tObject = "); Serial.print(mlx.readObjectTempC()); Serial.println("*C");
  Serial.print("Ambient = "); Serial.print(mlx.readAmbientTempF()); 
  Serial.print("*F\tObject = "); Serial.print(mlx.readObjectTempF()); Serial.println("*F");

  Serial.println();
  delay(500);
}
```
I'm following the exact example that comes with the library. If I upload the same sketch to other Arduino boards and connect just the SDA and SCL pins to the sensor, it works (the rest of the connections remain on the due). I've read that this might be an [issue  ](https://forum.arduino.cc/t/i2c-repeated-start/134401/16) with the adafruit and the arduino Wire library. I've checked the Wire library and it has since been changed to reflect this configuration, but the adafruit library has not been fixed. I tried the manual fixes but they are either not working for new versions of arduino IDE or I'm doing them wrong. 

The response I get from the sensor:
```
Adafruit MLX90614 test

Ambient = 1037.55*C	Object = 1037.55*C
Ambient = 1899.59*F	Object = 1899.59*F
```

And it doesn't change at all

caternuson, 2021-06-29:
> If I upload the same sketch to other Arduino boards and connect just the SDA and SCL pins to the sensor, it works

I2C on the Due in general is known to be not so great. Can you just use one of the other boards that works?

caternuson, 2021-06-29:
Try the 2.0.0 release and see if it helps. Maybe the BusIO usage will fix this also?
https://github.com/adafruit/Adafruit-MLX90614-Library/releases/tag/2.0.0

LEBsci, 2021-06-30:
Thank you @caternuson. I just tested the new release and sadly it still doesn't work. I was asked to use this board since there is a possibility that we will need the faster processor. We are still in early tests so I can not say for sure, but maybe the other boards I have (Mega and UNO) will not be fast enough for the project. That's why I was hoping to find a fix for this.

caternuson, 2021-06-30:
FWIW - there are numerous more modern 32 bit ARM based Arduino boards that are faster than the due and have better I2C implementations.