adafruit/Adafruit-MLX90614-Library #30: MLX90614 I2C communications on Arduino Nano Every using Wire.h and Adafruit library

Nolan-Walker, 2021-07-25:
Hi there,

I am trying to get my Melexis MLX90614-DCI (medical grade version) sensor working correctly; it works with the regular arduino nano but not the arduino nano every. I can use I2C scanner to check that the sensor is there and has address 0x5A but the communications are not working. When using the the Adafruit mlxtest example the serial console prints "nan" indicating that the read failed. I should note that when using a cheap amazon version of the MLX90614 and the Arduino Nano Every, the mlxtest example works fine.

After some research it appears that the lilbrary may not support the new Arduino Nano Every; any help would be greatly appreciated.

Nolan

caternuson, 2021-08-05:
If it were a library issue, should happen with any MLX90614. Are you using a 5V variant of the MLX90614?

Nolan-Walker, 2021-08-05:
Ya I thought this too but I'm absolutely stumped as to why it will work
with any other Arduino Nano and not the Arduino Nano Every... the version I
am using is the medical grade DCI version which is 3.3V

caternuson, 2021-08-05:
The Nano's are 5V logic level boards:
https://store.arduino.cc/usa/arduino-nano
https://store.arduino.cc/usa/nano-every
It may have something to do with the mixed logic level?

Nolan-Walker, 2021-08-06:
I don't think that has anything to do with it... also the sensor works with
regular Nano just not the Nano every... I'm using I2C as well and also
using pull up resistors on the lines that pull up to 3.3v. It all works
with the regular Nano and I've tried several Nanos and they all work (just
not the Nano Every)