adafruit/Adafruit-MLX90614-Library #46: Reading raw values is not possible

hacknus, 2024-01-30:
The raw values are stored in 16bit signed integers and can be read out. Please implement this in the library as well, currently only unsigned ints are implemented (temperature).

caternuson, 2024-01-30:
Can you provide more information about what is needed here. Is this basically a request to make `read16()` public?

hacknus, 2024-01-30:
`read16` returns an unsigned 16 bit integer. However in the datasheet it says:

> If the RAM is read, the data are divided by two, due to a sign bit in RAM (for example, TO1 - RAM address 0x07 will sweep between 0x27AD to 0x7FFF as the object temperature changes from - 70.01°C...+382.19°C). The MSB read from RAM is an error flag (active high) for the linearized temperatures (TO1, TO2 and Ta). The MSB for the raw data (e.g. IR sensor1 data) is a sign bit (sign and magnitude format)

This means we need to read the raw data and convert it to a sign and magnitude format (signed 16 bit integer).
Also, I believe that the error flag for the temperature values is also not implemented.

caternuson, 2024-01-30:
OK, still not sure what the request is then. The various `read` functions take care of all that converting.

hacknus, 2024-01-31:
Doesn't the `read16` just return the unsigned two's complement instead of the sign and magnitude format?
`return uint16_t(buffer[0]) | (uint16_t(buffer[1]) << 8);`
see [here](https://en.wikipedia.org/wiki/Signed_number_representations).

so we'd need a `read16_signed` method (private is okay, if there is would be a wrapper like for example `readRaw1` and `readRaw2`).

And I also noticed that the CRC is never verified when reading data.
So to put concrete requests:

- implement `readRaw1` and `readRaw2` which return signed 16 bit integers
- implement the reading of the errorflag for TO1, TO2 and TA
- implement verifying the crc for any data that is read